timvideos/HDMI2USB-numato-opsis-hardware #9: DisplayPort lanes need DC-blocking (AC coupling) capacitors

mithro, 2015-07-19:
The DisplayPort lanes and aux channel should have ~0.1uF (100nF) DC blocking capacitors between the connector and the FPGA.

The AC coupling capacitor must be in the range of 75 nF to 200 nF. A value of 100 nF is recommended. These capacitors also need to be as physically small as possible, a package size of 0201 is recommended but a 0402 is acceptable. They should be placed at the same length from the FPGA/connector.

mithro, 2015-07-19:
The capacitors where added to the schematic in 4ada452b0a6dd5e9f09a80e1f66faf846f8c2c75

mithro, 2015-08-18:
Here are a couple of designs which have the correct DC blocking capacitors;

* Atrix-7 based device; [image]

* Cyclone-V based device using a TI redriver;
[image]

The FPGA does have built-in AC coupling in the receiver, however the design guide recommends using external AC coupling for DisplayPort. See the following picture;
[image]

mithro, 2015-08-18:
From @anoopnumato 
> Is it really necessary to cut the traces and add the dc blocking capacitors on this board? Can you test current prototype without these Capacitors?

Missing the DC blocking capacitors will degrade performance, but I'm uncertain by how much (and believe it really depends on what the system is connected too). Using the internal AC coupling might (but also might not) be enough that performance is still okay for doing initial testing.

Cutting the traces and adding the capacitors can also reduce performance because the matching and grounding is no longer correct.

mithro, 2015-08-18:
From @anoopnumato 
> We have tested GTP of SPN602 with out these capacitors on its reference clock and we are able verify the speed up to 3Gbs using chipscope pro.

anoopnumato, 2015-09-15:
Updated the files.

mithro, 2015-09-20:
Looks good.

closed 2015-09-20 by mithro