timvideos/HDMI2USB-numato-opsis-hardware #18: Add mounting holes for expansion boards when using a right-angle riser.

mithro, 2015-07-20:
All expansion boards should follow the PCI-Express mechanical standard so they can be mounted in ITX style cases using standard risers. 

However when not using a case, it would be good to allow the expansion boards to be securely attached to the motherboard. This can be done with a standard right angle connector (such as http://www.amazon.com/gp/product/B00KZIAHW8) and some extra mounting holes in the motherboard.

mithro, 2015-10-10:
See the following two pictures
 * https://drive.google.com/file/d/0B5VlNZ_Rvdw6WVNSQkxoLVhRam8/view?usp=sharing
 * https://drive.google.com/file/d/0B5VlNZ_Rvdw6TWxhWnBKeF9sU0k/view?usp=sharing

anoopnumato, 2015-10-16:
@mithro It will be difficult to add , required much re-routing .

mithro, 2015-10-17:
@anoopnumato Please provide mechanical diagrams showing the issues. There is plenty of empty space on the Opsis were we can put standoff holes.

We control the TOFE specification, so we can put the standoff holes anywhere.

mithro, 2015-10-21:
@anoopnumato Why did you assign this back to me? Did you see my last comment?

mithro, 2015-11-02:
@anoopnumato is going to;

 * upload a couple of mechanical diagrams.
 * make sure their are suitable mounting holes for half-height cards.
 * (maybe) add the following (optional) mounting holes

[image]

commit fe1ce21f (2015-11-03): Adding extra mounting holes for Issue #18.

Fixes #18.
PR #40 (2015-11-03): Changes on 2015-11-03 3.50PM
Fixed #18
closed 2015-11-03 by mithro