timvideos/HDMI2USB-numato-opsis-hardware #24: Feature: VGA-in

tillsteinbach, 2015-08-15:
This is probably more a feature request for a future version:
Lots of speakers still expect a VGA display. So it would be great to still have one (additional) VGA input.

mithro, 2015-08-16:
@tillsteinbach We already have an expansion board for the Digilent Atlys board (our old prototyping board) that does VGA. You can see it at https://github.com/timvideos/HDMI2USB-vmodvga

This board was made by @rohit91 as part of Google Summer of Code. You can see his work at http://dreamsxtrinsic.blogspot.de/2014/08/gsoc-project-report-vga-capture.html

It would be pretty simple to port this board to the PCI-Express interface used by the Numato Opsis board but I don't think @rohit91 has any time to do that.

tillsteinbach, 2015-08-17:
Ahh! Nice, thanks for letting me know!

sealne, 2015-10-16:
I plan to work on converting the layout to TOFE while Rohit plans to focus on getting it to work with the new misoc based firmware

mithro, 2015-10-17:
@sealne and @rohit91 are working on this! Yay!

mithro, 2015-10-24:
https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/30

mithro, 2015-10-24:
FYI There is now a TOFE template available for KiCad at https://github.com/timvideos/HDMI2USB-TOFE-kicad-template

mithro, 2015-11-03:
I'm closing this issue as the project has it's own repo now.

closed 2015-11-03 by mithro