timvideos/HDMI2USB-numato-opsis-hardware #25: DisplayPort lanes on the sink are incorrect

mithro, 2015-08-26:
Compare https://bitguru.files.wordpress.com/2008/11/displayport_pinout11a.gif to 
[image]

mithro, 2015-08-26:
Potential fix? - http://www.xilinx.com/support/documentation/application_notes/xapp491.pdf

enjoy-digital, 2015-08-26:
http://www.xilinx.com/support/documentation/user_guides/ug386.pdf

"If the TXP/TXN differential traces are swapped on a board, tie TXPOLARITY High"
and
"If the polarity of RXP/RXN needs to be inverted, RXPOLARITY must be tied High"

So it should be OK.

mithro, 2015-08-31:
FYI - @anoopnumato @numato

mithro, 2015-11-02:
@anoopnumato has fixed this.

[image]

closed 2015-11-02 by mithro

mithro, 2016-01-04:
With a standard DisplayPort cable;

GTPA1_DUAL_X0Y0_0(TX) goes to GTPA1_DUAL_X1Y0_1(RX)
GTPA1_DUAL_X0Y0_1(TX) goes to GTPA1_DUAL_X1Y0_0(RX)
GTPA1_DUAL_X1Y0_0(TX) goes to GTPA1_DUAL_X0Y0_1(RX)
GTPA1_DUAL_X1Y0_1(TX) goes to GTPA1_DUAL_X0Y0_0(RX)