timvideos/HDMI2USB-numato-opsis-hardware #30: Create a KiCad template for TOFE expansion boards

mithro, 2015-09-24:
We need a template which has the following things;

 * The TOFE connector part in schematic form.
 * Example PCB which includes;
  * The PCB fingers.
  * The mechanical size for "full size PCI-Express card"
  * The mechanical size for "half size PCI-Express card"
  * Keep out zones.
  * Mounting holes.

See https://github.com/mithro/kicad-mini-pci-express for an example I did for mini-pci-express.

Useful information;
 * http://read.pudn.com/downloads166/ebook/758109/PCI_Express_CEM_1.1.pdf (see `6. Add-in Card Form Factors and Implementation`)
 * http://pcisig.com/specifications/pciexpress/base
 * http://www.interfacebus.com/PCIe_Card_Dimensions.html

We can then add some extra tools like;

 * The required EEPROM parts + tool to generate the EEPROM contents.

anoopnumato, 2015-10-16:
@mithro Please verify the files shared and close the issue

mithro, 2015-10-17:
The TOFE template is available at https://github.com/timvideos/HDMI2USB-TOFE-kicad-template 
Any issues with the template should be logged in that repository.

Closing.

closed 2015-10-17 by mithro