timvideos/HDMI2USB-numato-opsis-hardware #31: Mini-ITX case alignment issue for I/O shield

mithro, 2015-10-01:
The board has an alignment issue when amounted in an Mini-ITX case the I/O ports don't correctly line up with the I/O shield area.

mithro, 2015-10-01:
[image]
[image]
[image]

mithro, 2015-10-01:
It appears we have some type of alignment issue with our board. The DisplayPort connector on the far right is too close to the edge and there seems to be some type of gap on the left side. I see the same issue with the board in the case I got.

@anoopnumato , can you please verify the board is matching the MiniITX / ATX standard for the hole and connector positions? It kind of looks like the board should be shifted 5mm to the right?

anoopnumato, 2015-10-01:
@mithro As per the Mini-iTX spec. required spacing of the IO connector side is 6.35mm(page 9), in the attached image of Numato Opsis it was 7.2 mm.
[image]
http://www.formfactors.org/developer%5Cspecs%5CMini_ITX_Spec_V2_0.pdf

anoopnumato, 2015-10-01:
@mithro Seems it requires 7.52 and we have 7.32!

mithro, 2015-10-01:
What are your suggestions for fixing?

anoopnumato, 2015-10-01:
Option one is change the display port TX and RX connectors clearance 3 mm instead of 5 mm.

mithro, 2015-10-04:
My recommendation is that we do the following;
 * Move P17 around the corner so it is between the USB connector and the Cypress FX2
 * Move DisplayPort TX left a bit.
 * Move DisplayPort RX closer to DisplayPort TX.

Is that doable?

Why is there a gap on between the side of the board and the left most edge of the board? Can you please recheck all the following too:
 * The hole positioning
 * The expansion header positioning

mithro, 2015-10-21:
Distance was changed to ~13mm between connector edge and hole center.

closed 2015-10-21 by mithro