timvideos/HDMI2USB-numato-opsis-hardware #37: Noise on DisplayPort high speed lines when using the AUX connector

mithro, 2015-10-18:
When Mike tries to use the AUX channel on the DisplayPort connector, it causes the high speed link to drop. We are currently investigating;

From Mike;
<blockquote>
Whenever the link is up and stable:

* If I leave the AUX channel tri-stated then the link stays up.

* if I waggle the AUX channel to ask the sink for it's status I get the main links drop,

* There is high frequency 'noise' on the AUX channel when tristated 

* If I drive the AUX channel to 0 or 1 when idle the link doesn't come up.
</blockquote>

This is quite probably a side effect of https://github.com/timvideos/HDMI2USB-numato-opsis-hardware/issues/9 -- but we need to confirm.

mithro, 2015-10-18:
@hamsternz said;
<blockquote>
Tested for a short with a scope and couldn't see anything (watched the lanes while AUX EDID probes were going on).
</blockquote>

mithro, 2015-10-18:
As the repository has moved forward, I've uploaded an older version of the schematic for the preproduction boards at https://github.com/timvideos/HDMI2USB-numato-opsis-docs/blob/gh-pages/pdfs/NumatoOpsis-schematic-v2-prototype-13-gb1c4312.pdf

mithro, 2015-10-18:
Oh wait, that version is still too new. It has the decoupling capacitors. Will need to go back further.

hamsternz, 2015-10-18:
Here's the change H/W 8b/10b makes in resource usage.

Single channel 800x600 colourbars
            H/W   S/W
Registers  724   878
LUTS       882  1397
Occupied   295   552

Four channels 800x600 white
            H/W   S/W
Registers  820   975  -16%
LUTS       940  1473  -36%
Occupied   320   562  -43%

It is all checked in now, and It should be all good for you to pull it.

Mike