timvideos/HDMI2USB-numato-opsis-hardware #59: Confused about HDMI resolution

nesteruk, 2016-12-06:
When I watched the [interview](https://www.youtube.com/watch?v=b0JEVifDHQ0), you mention that HDMI only supports 720p. But on the [website](http://numato.com/numato-opsis-fpga-based-open-video-platform/), we see:

> HDMI:
>                     Inputs: 2 
>                     Outputs: 2
>                     Resolution: 1080p30, 720p60, 1280×1024@60Hz

So which is it? 1080p or 720p?

mithro, 2016-12-06:
1080p normally means 1080p60 (1080p at 60 frames per second). The opsis is currently only capable of 1080p30 which while is the same resolution, is only half the frame rate.

There is a bunch more information at https://opsis.hdmi2usb.tv/info/video-info-faq.html

Does that help?

Does that help?

nesteruk, 2016-12-06:
@mithro Ahh, I get you, but I'm not sure why the FPS specification would be important -- the Opsis is positioned at an event recording/streaming device. We never do it at 60fps. I mean, typical video recording is done at 30fps, screens are recorded at 30fps _at most_ (and for coding, you don't need anything above 15fps), and certainly nobody would realistically be streaming at anything above 30fps. Am I missing something here?

mithro, 2016-12-06:
Many grumpy projectors out there only support 1080p60 and not 1080p30.

As well, if I brought a device which claimed to be "1080p" but is didnt support 1080p60, I would be pretty frustrated.

Lastly, we don't test with 1080p30 frequently (and support is not enabled by default) so it seems wrong to claim it out of the box.

mithro, 2016-12-06:
Oh, if it is not clear this is the maximum resolution that can be passed through the opsis. You can't use 1080p60 to your projector and capture at 1080p30.

mithro, 2016-12-06:
Btw You should come join us on IRC in the #timvideos channel on irc.freenode.net (web version at https://webchat.freenode.net/?channels=#timvideos ).

nesteruk, 2016-12-06:
Interesting. Did not that projectors can be that picky. That said, I've had _a ton_ of issues with projectors for the conferences we do. It's been a real pain because you rely on the venue to provide these.

Anyways, I think I'm going to go ahead and close this issue.

closed 2016-12-06 by nesteruk

mithro, 2016-12-06:
@nesteruk - Please do email me at mithro@mithis.com! I would love to find out what you do, how you found out about the Opsis and how we might collaborate.

The TimVideos IRC channel also has a bunch of people from different open source groups who do conference recording and capture (PyCons, DebConfs, Chaos Club, etc). There is always a bunch of discussion about how to solve different problems, what hardware to use, etc. It is a great place to hang out if you care about this topic.

mithro, 2016-12-06:
(You can also find out about cool new things in development :-)