timvideos/HDMI2USB-numato-opsis-hardware #60: pictures of the Opsis board

CarlFK, 2017-09-20:
This is not the same as the board I have:
https://github.com/timvideos/HDMI2USB-numato-opsis-hardware/blob/master/img/opsis-v2-prototype.jpg

2 threaded connectors, (composite?) different power connector 

also add: top, bottom, all 4 sides.

mithro, 2017-09-20:
Pictures are at https://opsis.hdmi2usb.tv/getting-started/overview.html

mithro, 2017-09-20:
And https://opsis.hdmi2usb.tv/getting-started/layout.html

CarlFK, 2017-09-20:
reopening because the picture I referenced is still incorrect - note the file name is "prototype" 
https://github.com/timvideos/HDMI2USB-numato-opsis-hardware/blob/master/img/opsis-v2-prototype.jpg

also  prototype:  https://opsis.hdmi2usb.tv/  https://opsis.hdmi2usb.tv/img/numato-opsis.jpg

If this isn't the place to find pictures, delete it the whole dir. 
https://github.com/timvideos/HDMI2USB-numato-opsis-hardware/tree/master/img

There are no pictures/diagrams showing the orientation of the plugs.   Like is the rj45 release clip on the top or bottom?

the picture of the 6 pin power plug is hard to know how it is keyed. 
https://opsis.hdmi2usb.tv/img/highlights/Jumpers.jpg

mithro, 2017-09-20:
https://github.com/timvideos/HDMI2USB-numato-opsis-docs/tree/master/img

CarlFK, 2017-09-20:
That is the Prototype. Or whatever it is, it does not show the Six pin
power connector.

mithro, 2017-09-20:
All the possible information you want on the power supply should be here -> https://opsis.hdmi2usb.tv/getting-started/power.html

Including details on the six pin connector keying.

CarlFK, 2017-09-20:
This bug report is about the picture(s) being incorrect. 

There is also a bug report about connector docs, which I found confusing.   A picture of the production board might help.